neohere97/SkySync #17: Component Library & Footprint Design

neohere97, 2023-09-24:


mrya3714, 2023-09-29:
Schematic refernce : [https://www.silabs.com/documents/public/schematic-files/BRD4184A-A01-schematic.pdf](url)

neohere97, 2023-09-29:
@akash7071 Designator is on Top Layer and Pads need to be rounded rectangle ?

neohere97, 2023-09-29:
AS7331 Currently finished but needs cross verification, USB C imported from SnapEDA needs cross verification

mrya3714, 2023-09-30:
PMIC Buck converter Inductor 10uH: [https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/744029100/1638496?s=N4IgTCBcDaIOwBYEAYwE4CMzkAIQF0BfIA](url)

mrya3714, 2023-09-30:
PMIC Boost converter Inductor 22 uH : [https://www.we-online.com/components/products/datasheet/744031220.pdf](url)

neohere97, 2023-10-01:
https://www.digikey.com/en/products/detail/jst-sales-america-inc/BM02B-SRSS-TB/926694 - JST Micro Connector

neohere97, 2023-10-01:
https://www.digikey.com/en/products/detail/anysolar-ltd/SM141K06L/9990462
Solar Panel

closed 2023-10-10 by neohere97